chenxuuu/sms_forwarding #1: 长短信未合并问题

4ft35t, 2022-11-21:
1. 长短信未合并
2. 长短信是倒序
[image]

chenxuuu, 2022-11-21:
1. 因esp-idf库限制，get方式长度最长仅为2048字节（一个utf8汉字经过url编码后为9字节），如需一次性发送，请使用post方式
2. 此为平台问题，请询问平台。设备端为顺序发送

closed 2022-11-21 by chenxuuu

4ft35t, 2022-11-21:
确认是 POST 方式发送，只修改过 notify.lua 中的发送地址，及添加发送方号码
```
                local data = pdu.ucs2_utf8(sms[2])
                log.info("notify","send to wepush",data)
                code, h, body = http2.request(
                        "POST",
                        "http://192.168.1.123:8000/cgi-bin/wepush.sh",
                        {["Content-Type"] = "application/x-www-form-urlencoded"},
                        "d=from:"..string.urlEncode(sms[1]).." "..string.urlEncode(data)
                    ).wait()
                log.info("notify","pushed sms notify",code,h,body,sms[1])
```

chenxuuu, 2022-11-21:
看了你上面的截图，看起来这一条`长短信`不同的部分居然花费了三分钟，应该是代码超时认为是三条独立的短信了

我认为这个问题不属于代码应处理的范畴

4ft35t, 2022-11-21:
这个确实时间有点久了，遇到新的再来。

36h, 2022-11-22:
参考教程自己尝试了一下，联通卡短信是分开推送的。采用的 LuatOS社区提供的推送服务的BARK推送
串口的日志 [image]

BARK推送[image]

36h, 2022-11-22:
看到了 谢谢

stevenliuit, 2023-01-30:
确实 get请求超时 提示
[2023-01-30 23:09:25.113] I/http HTTP GET
[2023-01-30 23:09:25.118] [0;31mE (54922) HTTP_CLIENT: Out of buffer[0m
[2023-01-30 23:09:25.122] D/http esp_http_client_perform -1
[2023-01-30 23:09:25.126] I/user.notify	pushed sms notify	-1	nil	nil
esp-idf库限制 好恶心啊

chenxuuu, 2023-02-20:
最新版本LuatOS的ESP32C3的HTTP组件，已被LuatOS主仓库的HTTP库替换，最新固件已无GET长度限制

脚本也已做出相应改动

4ft35t, 2023-03-14:
短信合并问题依旧存在，联通卡。

[image]

我在用的解决思路，我只需要收短信，不管短信长短，收到就转发，不处理合并逻辑。

直接删除 main.lua 91-118 行即可。

https://github.com/chenxuuu/sms_forwarding/blob/master/script/main.lua#L91-L118

有遇到同样问题的朋友可以试一下。

pengnb, 2023-03-20:
我是这样改的，测试了两条合并没问题，其他没试过。我是推送到自己群晖的synology chat

    while true do
        collectgarbage("collect")--防止内存不足
        local _,phone,data,time,long,total,id = sys.waitUntil("AT_CMT")
        if long then--是长短信！而且是第一条			
            log.info("air780","found a long sms",total,id)
            --缓存，长短信存放处
			local smsTemp = {}
			for i = 1, total do
				smsTemp[i] = "" -- 为 table 的每个元素设置默认值
			end
            smsTemp[id]=data
            --发件人
            local lastPhone = phone
            --等待下一条
            for i=1,total do
                local r,phone,dataTemp,time,long,total,id = sys.waitUntil("AT_CMT",10000)
                if r then
                    if phone == lastPhone then
                        log.info("air780","a part of long sms",total,id)
                        smsTemp[id]=dataTemp
                        --收到完整的长短信
						if #smsTemp == total then
							data = table.concat(smsTemp)
							log.info("air780","got a long sms")
							break
						end
                    else--手机号不一样了？那就是两条不同的短信了
                        data = table.concat(smsTemp)--这是长短信的内容
                        log.info("air780","a long sms cut")
                        notify.add(phone,dataTemp)--这次的短信
                        break
                    end
                end
            end
        end
        notify.add(phone,data)--这次的短信
    end

chenxuuu, 2023-06-25:
已为乱序长短信特殊处理