chenxuuu/sms_forwarding #7: 发现存跑流量的情况，不过比较少就几KB

zhouwenze, 2023-03-14:
## 描述一下这个bug / Describe the bug
请使用简介并详细的语句，来描述这个bug。 / A clear and concise description of what the bug is.
使用Air780E+ESP32C3方案后，发现系统有在跑流量，不过只有几KB

## 复现步骤 / To Reproduce
按照下面的步骤，可以复现bug / Steps to reproduce the behavior:
1. 插卡，上电，开机运行，过了一天，发现扣了一些流量费用
 
## 预期的行为 / Expected behavior
清楚简洁地描述你希望发生的事情。 / A clear and concise description of what you expected to happen.
希望可以不走流量

## 截图 / Screenshots
如果需要，请放上你的截图。 / If applicable, add screenshots to help explain your problem.
暂无。

## 日志 / Logs
`更多设置`--`打开串口收发日志目录`，可以看到日志 / Upload your log files.
暂无。

## 系统信息 / Desktop (please complete the following information):
 - OS: [e.g. iOS]
 - Version [e.g. 22]
 - 如果可以，可以附加配置文件内容（`更多设置`--`打开串口收发日志目录`--切换到上一级目录--`settings.json`文件）

## 其他备注信息 / Additional context
Add any other context about the problem here.

chenxuuu, 2023-03-14:
可以去官方仓库提bug

chenxuuu, 2023-03-17:
新代码看会不会有类似现象，还有的话就去官方提bug

chenxuuu, 2023-06-08:
还会消耗流量吗

chenxuuu, 2023-06-14:
实测还会消耗，我抓日志看看at做了什么操作

wendal, 2023-06-16:
是每天的OTA请求吧

chenxuuu, 2023-06-16:
@wendal 并不是，是因为[这个](https://gitee.com/openLuat/airm2m_ec618_at/issues/I7DYJN)

chenxuuu, 2023-06-26:
最新版的AT固件已无此现象，更新到今天这条评论之后发布的AT固件版本即可

closed 2023-06-26 by chenxuuu

antonchen, 2023-07-13:
> 最新版的AT固件已无此现象，更新到今天这条评论之后发布的AT固件版本即可

请问是指 [V1141](https://gitee.com/openLuat/airm2m_ec618_at/releases/tag/AirM2M_780E_LTE_AT_V1141) 版之后的固件吗？这个版本我下载不了，不知道什么情况