chenxuuu/sms_forwarding #13: 申请单独添加Bark推送功能

PandaTAT, 2023-09-23:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.

由于使用其他第三方平台推送还需要中转一下不太方便也不安全，希望能直接在脚本中添加bark推送功能，这样也能自建后端服务器使用自己服务器完成推送

### 描述您想要的解决方案 / Describe the solution you'd like

直接在代码中直接使用bark推送官方api进行短信转发服务

### 描述您想要的详细使用步骤描述 / Describe the solution you'd like to use in what way

在config中添加bark token和推送服务器连接配置功能，并在notify.lua中添加bark推送请求json代码以便直接使用bark推送服务

### 其他备注信息或截图 / Add any other context or screenshots about the feature request here

自己尝试通过在server酱的代码基础上进行修改但是没有成功，以下是修改的内容：
[image]

下面是报错内容：
[image]

### 确认信息

- [X] 检查过该需求，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 这个需求是真实存在的，并且对于本项目来说是必要的
- [X] 我已经提供了该需求的清晰、详细、完整的方案，请按该方案进行实施
- [X] 我认为我提供的信息已经足够，不需要其他人继续完善该需求的其他细节
- [X] 我同意跟进该需求的实施进程，并在适时提供合理的解释与纠正

keiwo, 2023-11-03:
@PandaTAT 
26行，local luatosPush = "Bark那串代码"，
132行，local url = "http://你的自建域名/"..luatosPush.."/"..sms[1].."/"..data

我现在遇到的问题是无法使用https。

chenxuuu, 2023-11-13:
受空间所限，目前固件不支持强制https，需要服务端进行更改适配

chapiom, 2023-12-07:
bark用的是get不是post，把下面luatos官方推送的url换一下就可以用了。
tls问题我也碰到了，把固件从1005版本换成1004就正常了，我也不知道为什么。

chapiom, 2023-12-07:
另一个项目找到了答案，官方编译的1005把内存用完了，精简版的就可以。

maokwen, 2023-12-11:
bark 你需要用 v2 api，只有 v2 是支持 post 的

closed 2024-03-04 by chenxuuu