chenxuuu/sms_forwarding #21: 请问是否有可能支持30X转发

terrytw, 2024-11-30:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.

原来用自建的一个推送服务, 后来域名改了, 设置了域名跳转, 但是似乎本脚本不会自动重新请求新域名

### 描述您想要的解决方案 / Describe the solution you'd like

```
                        code, h, body = http.request(
                                "POST",
                                "https://api.pushover.net/1/messages.json",
                                {["Content-Type"] = "application/json; charset=utf-8"},
                                json_body
                            ).wait()
```
举例来说上述代码中`api.pushover.net`, 如果它设置了跳转到`api2.pushover.net`, 如果能自动跟随发出新请求就好了

### 描述您想要的详细使用步骤描述 / Describe the solution you'd like to use in what way

N/A

### 其他备注信息或截图 / Add any other context or screenshots about the feature request here

N/A

### 确认信息

- [X] 检查过该需求，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 这个需求是真实存在的，并且对于本项目来说是必要的
- [X] 我已经提供了该需求的清晰、详细、完整的方案，请按该方案进行实施
- [X] 我认为我提供的信息已经足够，不需要其他人继续完善该需求的其他细节
- [X] 我同意跟进该需求的实施进程，并在适时提供合理的解释与纠正

chenxuuu, 2024-12-01:
手动判断返回值及其header内容

 

 
 
 
 
 
 
 
  
 
 
 
 
  
&mdash;

terrytw, 2024-12-01:
好的, 多谢解答

closed 2024-12-01 by terrytw